bluerobotics/cockpit #555: Provide a "device configuration protocol" for widgets

ES-Alexander, 2023-11-09:
It's valuable to have consistent UI elements, so I think it could be handy to have a defined protocol that widgets/services can use to share a device configuration API, which Cockpit then creates the UI for from the relevant field types and descriptions, and presents that when the user configures the corresponding widget.

I thought of this [here](https://github.com/bluerobotics/BlueOS/issues/2187), in the context of supporting Ping family sonar devices, and realised it could be extra useful if we did it as a general thing. That was the widget designer can focus on primarily the UI for the main display, and an API for the relevant functionalities, and then Cockpit can handle the UI for configuration stuff.

rafaellehmkuhl, 2023-11-09:
Challenging, but very powerful.
We should discuss more about this.

patrickelectric, 2023-11-10:
It can do something based on how MCM exposes v4l controls, it'll almost the same thing for onvif as well.

ES-Alexander, 2023-11-10:
> It can do something based on how MCM exposes v4l controls, it'll almost the same thing for onvif as well.

Bonus points if we can use the same system/protocol for all three :-)