bluerobotics/cockpit #560: Stabilizing the Raspberry Pi

itaru-fujitsuka, 2023-11-13:
Turn off Wifi
$ sudo iwconfig wlan0 txpower off

Turn off Bluetooth
$ sudo rfkill block bluetooth

Stop USB devices
$ sudo modprobe -r usbcore

...etc.

Stabilize the Raspberry Pi by reducing processing and power consumption

rafaellehmkuhl, 2023-11-13:
Hi @itaru-fujitsuka!
If I understand correctly, those are suggestions on how to improve the performance of Cockpit when running through BlueOS, right?

itaru-fujitsuka, 2023-11-13:
Hi @rafaellehmkuhl !
In our team, the mechanic and the pilot are different individuals.
While the mechanic may also act as a pilot, the pilot's role is solely focused on operations.
The mechanic is responsible for configuring BlueOS and setting up Cockpit.
However, the pilot only uses Cockpit for operating BlueROV2.

To ensure the stability of BlueROV2 and Cockpit, there is a need to switch the Raspberry Pi settings from Cockpit.
This is particularly crucial for reducing the power consumption of the Raspberry Pi, preventing thermal issues, and extending the operational time underwater.

joaoantoniocardoso, 2023-11-18:
> Hi @rafaellehmkuhl ! In our team, the mechanic and the pilot are different individuals. While the mechanic may also act as a pilot, the pilot's role is solely focused on operations. The mechanic is [...]

Hi @itaru-fujitsuka,

do you mind sharing the reasoning behind turning off Wifi/Bluetooth/USB?

Thanks

itaru-fujitsuka, 2023-11-20:
Hi @joaoantoniocardoso .

Firstly, when we were using the BlueROV2 underwater this summer, it would become very unstable after about 20 minutes.
After conducting various tests on land, we suspected overheating issues with the camera and Raspberry Pi.

As a result, we opted for an IP camera to reduce the load on the Raspberry Pi.
Despite turning off WIFI from BlueOS to review detailed settings, there was no reduction in power consumption.

However, when turning off WIFI through commands, power consumption decreased by around 0.5-1A, and stability improved.
This led us to the idea of turning off unnecessary features and we would like this implemented.

rafaellehmkuhl, 2023-11-20:
> Hi @joaoantoniocardoso .
>
> Firstly, when we were using the BlueROV2 underwater this summer, it would become very unstable after about 20 minutes. After conducting various tests on land, we suspected [...]

This is very interesting.
@Williangalvani haven't you experienced the same problem once?

Williangalvani, 2023-11-20:
> This is very interesting.
@Williangalvani haven't you experienced the same problem once?

I don't recall anything similar, not really. But this is worth investigating. I don't think it should be a feature of Cockpit per se, but it would be nice if we had arbriraty commands with arbitrary code for people wanting to do this =]

patrickelectric, 2023-11-20:
> I don't recall anything similar, not really. But this is worth investigating. I don't think it should be a feature of Cockpit per se, but it would be nice if we had arbriraty commands with arbitrary [...]

related to: https://github.com/bluerobotics/cockpit/issues/577