bluerobotics/cockpit #561: ROV status warning function

itaru-fujitsuka, 2023-11-13:
Smaller display with icon when all clear.
Increased display when a warning occurs.

-Leak Sensor
-Raspberry Pi status
 Temperature, CPU usage, memory
-Battery status
 Voltage, current, remaining capacity
-Communication status
-Pressure
etc.

rafaellehmkuhl, 2023-11-13:
Hi @itaru-fujitsuka!
Could you detail this issue a little more?
Is it about the `VeryGenericIndicator`?

itaru-fujitsuka, 2023-11-13:
Hi @rafaellehmkuhl !
The feature to change the color of the icon based on thresholds, similar to QGC, is planned for VeryGenericIndicator.
I think it would be beneficial to have the option to hide based on color.
In other words, I would like the icon to be displayed when it exceeds the configured threshold.
(It would be convenient if default values for commonly monitored parameters were bundled together, making configuration easier.)

rafaellehmkuhl, 2023-11-13:
Cool! This is a nice idea indeed.

I will change the title of the issue so it better reflects your idea, ok?

ES-Alexander, 2023-11-29:
Nice idea! :-)
This is a limited-domain expansion of #49.

It may also be worth considering some of the ideas presented in #343, in a more general sense.