bluerobotics/cockpit #562: Modes for operators of tether

itaru-fujitsuka, 2023-11-13:
Consider the requirements.

rafaellehmkuhl, 2023-11-13:
Hi @itaru-fujitsuka!
Could you explain a little bit better what this issue means?

itaru-fujitsuka, 2023-11-13:
Hi @rafaellehmkuhl !
In QGC, only one display is possible, so currently, both the tether operator and the captain see the same screen as the pilot.
However, Cockpit is fantastic because it supports dual pilots.
This means that it can provide screens displaying the information needed for each role.

Therefore, I thought it would be good to organize information not only for the pilot's role but also for others performing different tasks, making it selectable by default.

rafaellehmkuhl, 2023-11-22:
From what I understood, the Captain would be using a computer, and the Theter operation using another, so they would be running different Cockpit profiles, right?

In this case, is there something missing in Cockpit for that or something that would help to improve the experience for this use case?