bluerobotics/cockpit #574: Mode Selector Component Not Showing Current Mode

ericjohnson97, 2023-11-19:
It seems that the the mode selector component is able to command mode changes, but goes blank after a mode change is commanded.  

I think this is because the 
```
const mode = ref<string | undefined>(undefined)
```
is not being driven by anything. I think this should be driven by the heartbeat message.

rafaellehmkuhl, 2023-11-19:
Eric, I'm not being able to reproduce the issue mentioned.
In which case are you getting it?

ericjohnson97, 2023-11-19:
@rafaellehmkuhl After further inspection of the code I think sub and rover update the mode variable, but plane and copter do not.   

in sub I see this line 

https://github.com/bluerobotics/cockpit/blob/f9a899abb152640c447b9f1ec5179395a5578dc4/src/libs/vehicle/ardupilot/ardusub.ts#L98C51-L98C51

which this looks like a cleaner way to get the mode string than my function. However, it probably make sense to move this to the ardupilot.ts file since it is common across all vehicle types

rafaellehmkuhl, 2023-11-19:
> @rafaellehmkuhl After further inspection of the code I think sub and rover update the mode variable, but plane and copter do not.
>
> in sub I see this line [...]

Exactly that!

And I agree we should move the code, yes.
It's just a matter of consuming from the right `CustomMode`, which changes for each firmware.

ericjohnson97, 2023-11-20:
After looking at the code for a bit I am not sure how to actually move this to the ardupilot file. Maybe in the short term I can just implement the mode updating for plane and copter in their respective files as they are in sub in rover.  

As a heads up I won't have much time in the short term to make these changes.

rafaellehmkuhl, 2023-11-20:
I agree. This gives us time to decide on which is the best approach vehicle-wide for the future.

ericjohnson97, 2023-11-20:
closed with https://github.com/bluerobotics/cockpit/pull/575

closed 2023-11-20 by ericjohnson97