bluerobotics/cockpit #583: Inform the user that more than one IP is being used for video stream routing, encouraging to choose one

rafaellehmkuhl, 2023-11-22:
We've implemented the possibility of choosing the IP to be used for video stream routing on #571, so the functionality is already there, but 99% of the users won't know it is there, and neither that it can solve their video stuttering problem.

My proposal is that, when the user hasn't yet choose an IP route, and there's more than one available, that we open a popup informing about that and encouraging to choose one of the IPs for routing (usually the one for the cabled network interface).

This is important so we don't have users with the video stuttering problem without knowing it already can be solved from their side.

joaoantoniocardoso, 2023-11-22:
To help the user decision, for now,
```
curl -X 'GET' \
  'http://blueos.local:9111/v1.0/services' \
  -H 'accept: application/json'
```

would tell us exactly what kind of interface each IP responds to.

rafaellehmkuhl, 2023-11-22:
> To help the user decision, for now, `http://blueos.local:9111/v1.0/services` would tell us exactly what kind of interface each IP responds to.

Forgot about it! Yes, we should include this info!

PR #591 (2023-11-27): Warn user when multiple IP routes are being used for the WebRTC streaming
[image]

Fix #583
closed 2023-11-27 by rafaellehmkuhl