bluerobotics/cockpit #594: Eventing

shgavman, 2023-11-27:
Hi Rafael,

I suspect a lot of us require an eventing tool in our navigation software - this is needed when carrying out inspection.
A button can be assigned to logging an event. 
Pressing that button should capture all available telemetry, a screenshot (either the full screen including other devices such as sonar or only the camera window(s), and operator comment. 
It should be possible to export the data and ideally export a report. 

Suggested tables and fields:
Project Table
Project ID
Project Name
Client Name
Location
Start Date
End Date
Project Manager
Status

Dive Log Table
Dive ID
Project ID
ROV Pilot Name
Dive Date
Dive Time
Depth: Maximum depth reached.
Location: Specific location of the dive within the project area.
Purpose: Purpose or objective of the dive.
Conditions: Notes on weather and underwater conditions.
Equipment Used: List of equipment used during the dive.

Anomaly Log Table
Anomaly ID
Dive ID: Associated dive identifier.
LAT
Long
Depth
Location: Description of location if co-ords unavilable
Description: Detailed description of the anomaly.
Severity: Severity rating of the defect (e.g., Minor, Moderate, Severe).
Date Logged: Date when the anomaly was logged.
Photos/Videos: Links or references to associated visual documentation.
Action Required: Immediate action required, if any.
Action Taken: Details of any action taken to address the anomaly.
Status: Current status of the anomaly (e.g., Open, In Progress, Resolved).

Perhaps a test table to for Cathodic Prection Probes or Ultrasonic thickness gauge for example. That could come later. I think the above would be a good minimum viable product.

rafaellehmkuhl, 2023-11-27:
Do you think, on the event triggering (e.g.: joystick button click) we should log only one point or have several points during a period of time (e.g.: when the button is clicked we bring to the report a set of 10 telemetry points around this timestamp, including points before the button click, so the user can have data about events that just happened)?

ES-Alexander, 2023-11-30:
This idea is definitely useful, but I think it likely works better as an [add-on](https://github.com/bluerobotics/BlueOS/issues/1939) rather than part of the core Cockpit functionality, since 
- it's not required to operate a vehicle effectively
- it's only useful to a subset of users
- different users may want different fields and note types recorded, among other possibilities for future expansion, so a good implementation may end up quite complicated

I think it's important to set a precedent of minimising complexity and bloat in Cockpit for users who don't need it - the core should be a lightweight but powerful general platform, which users and developers can then build upon and extend to suit their more specific applications.

ES-Alexander, 2023-11-30:
#609 would be relevant to this, I think, although we may also need to introduce the concept of a hide-able widget (since we'd want some interface to add notes to when events are triggered, but it should then go away during normal operation).

rafaellehmkuhl, 2026-11-13:
Extra requirements raised on the [digital logbook](https://discuss.bluerobotics.com/t/an-integrated-digital-logbook-would-be-a-stand-out-feature/22804) forum thread that this issue doesn't spell out yet:

- Export the dive log as a formatted, client-ready PDF, or an editable format such as a spreadsheet, with a timestamp on every entry. Raw sensor data stays out of it.
- Log events automatically: power on and off, arm and disarm, and dive start and end (from a button in Cockpit or a joystick button).
- Show notes live on the video overlay.
- Programmable note buttons whose text can include a data-lake or MAVLink value (e.g. via the existing `{{ variableId }}` templates).