bluerobotics/cockpit #617: Joystick: suggested interface improvements

ES-Alexander, 2023-12-05:
[image]

[image]

- [ ] MAVLink/Cockpit/modifier icons for joystick functions
- [ ] Intuitive button icons for importing and exporting joystick and functions mappings
    - Clicking these should open a dialogue that allows choosing whether to import/export from/to the vehicle's onboard computer or the device running the Cockpit frontend (relevant to #556)
- [ ] Support modifier keys (advanced version of #453)
    - first 4 modifier keys are called `SHIFT`, `ALT`, `CTRL`, and `META`, and after that they're just numbered (or disallowed?)
    - [ ] Display tabs for modifier keys
        - pressing a button assigned to a modifier key moves to that tab, but tabs can also be changed by clicking
    - [ ] Functions mapping table for advanced configuration (/if there isn't an SVG)
        - columns are modifier keys, rows are (virtual) joystick button numbers
        - each column can be optionally counted as a MAVLink `MANUAL_CONTROL` shift button, except the REGULAR column
        - pressing a button/modifier-combination highlights the corresponding table cell(s), as well as the column+row headings
        - clicking a table cell allows changing the assigned function
        - if a cell is assigned as a modifier key, the rest of the row is greyed out
        - MAVLink `MANUAL_CONTROL` button function assignments also show the `BTNn` "n" number they're assigned to
    - adding a new modifier key is possible by clicking the `+` button in the tabs or on the right of the table
- [ ] Support multi-joystick prioritisation
    - show as tabs on the side
    - allow dragging joysticks to prioritise
        - by default only the top priority joystick has `MANUAL_CONTROL` outputs enabled (see #615)
        - if multiple joysticks enable `MANUAL_CONTROL`, outputs are determined based on prioritisation (e.g. both can press buttons, and control axes, but if one they're fighting over the state of a particular button function or axis then the higher priority one wins)

rafaellehmkuhl, 2023-12-05:
Those are some really good ideas.
I will implement at least the tabs and import/export buttons for now.

I specially love the icons in the functions names. They turn everything much more intuitive. And I'm fine with the `%`.

ES-Alexander, 2023-12-15:
Note: @Williangalvani is working on some potentially related/overlapping UI elements in BlueOS focused on sensor+peripheral calibration and configuration. We should probably try to create similar UIs so the two applications feel consistent/related.

rafaellehmkuhl, 2023-12-15:
> Note: @Williangalvani is working on some potentially related/overlapping UI elements in BlueOS focused on sensor+peripheral calibration and configuration. We should probably try to create similar UIs so the two applications feel consistent/related.

Can you get some examples?

ES-Alexander, 2023-12-15:
> Can you get some examples?

You’ll have to ask Will - I was talking to him about it, so I’m not sure if the code is committed/public yet.