bluerobotics/cockpit #623: Allow users to select their vehicle in the network

rafaellehmkuhl, 2023-12-06:
If for some reason the vehicle is not using ID 1, nothing is going to work.

rafaellehmkuhl, 2023-12-06:
That happened to Tony during his tests with some equipment that took ID 1 automatically.

joaoantoniocardoso, 2023-12-06:
> If for some reason the vehicle is not using ID 1, nothing is going to work.

I think even MCM expects 1

rafaellehmkuhl, 2023-12-07:
> > If for some reason the vehicle is not using ID 1, nothing is going to work.
> 
> I think even MCM expects 1

If I remember correctly, BlueOS is also completely hardcoded for vehicle in ID 1.

Williangalvani, 2023-12-09:
> > > If for some reason the vehicle is not using ID 1, nothing is going to work.
> > 
> > 
> > I think even MCM expects 1
> 
> If I remember correctly, BlueOS is also completely hardcoded for vehicle in ID 1.

BlueOS has some support for changing/using different vehicle IDs. It is useful for people with fleets of boats

joaoantoniocardoso, 2023-12-09:
Yep, MCM needs to be adjusted as well ([issue](https://github.com/mavlink/mavlink-camera-manager/issues/327))

rafaellehmkuhl, 2023-12-11:
> BlueOS has some support for changing/using different vehicle IDs. It is useful for people with fleets of boats

Interesting, didn't know that.
Where do I change the vehicle's ID?

goasChris, 2024-02-05:
Just going to add my voice onto the issue. I need to able to use a non `1` system_id in order for our usecase to work, and we would very much like to get to testing cockpit in our fleet when it becomes possible.

goasChris, 2024-02-06:
Please let me know if you want any help / testers, and I'll happily do so. I am not familiar enough with the cockpit code yet to be of much help writing the implementation, but I can at the very least provide some decent feedback :)

rafaellehmkuhl, 2024-02-06:
Thanks, Chris! I will ping you on that once we have the PR for testing.

joaoantoniocardoso, 2024-02-07:
> Yep, MCM needs to be adjusted as well ([issue](https://github.com/mavlink/mavlink-camera-manager/issues/327))

MCM (at git master) accepts a CLI argument to configure the mavlink system id :)

goasChris, 2024-02-21:
> Thanks, Chris! I will ping you on that once we have the PR for testing.

I was faster 🙈 

- [x] https://github.com/bluerobotics/cockpit/pull/740
- [ ] https://github.com/bluerobotics/cockpit/pull/754

With these two we would have solved most of the issues mentioned in this issue, I believe.
`Regarding MCM BTW, if the person uses Cockpit, the Mavlink ID for the MCM would be irrelevant.` https://github.com/bluerobotics/BlueOS/issues/2396#issuecomment-1953011024 , so that might not really be a problem?

I've not gotten around to testing cameras reliably, so I can't personally comment yet.

rafaellehmkuhl, 2024-02-21:
Nice!

Yes, the ID used in MCM is irrelevant for us using WebRTC, so this one is finished!

Thank you so much for the contribution Chris!

closed 2024-02-21 by rafaellehmkuhl