bluerobotics/cockpit #633: [Video Recorder] Fail to record multiple video streams

joaoantoniocardoso, 2023-12-09:
Two detected issues when trying to record multiple video streams:

1. If a second Video Recorder MiniWidget is added while the first is in recording mode, it interrupts the recording and it seems that the first widget goes to a weird state.

2. An error "Cannot load media stream" appears, shown in the video below:

[](https://github.com/bluerobotics/cockpit/assets/5920286/7c0aa1d8-d699-41aa-be1e-e6517bcd9f62)

joaoantoniocardoso, 2023-12-22:
Update after #654:

The error "2" is fixed, but the error "1" remains -- the recordings stop if you add another recording widget:

https://github.com/bluerobotics/cockpit/assets/5920286/cb219464-4e5a-46ad-afea-e7f094c8cf8d

rafaellehmkuhl, 2023-12-22:
> Update after #654:
> 
> The error "2" is fixed, but the error "1" remains -- the recordings stop if you add another recording widget:
> 
>  recordings_stop_if_another_record_widget_is_added.mp4

I expect that to be fixed on #655.

PR #655 (2023-12-23): Improve video recording management
Put the video store in charge of managing the video recordings, so recording information gets stored with the stream data. There are two main benefits on this:
- The lifecycle of a widget (e.g.: a view being changed and a recording widget being unmounted) does not affect the state of a recording
- Multiple recordings, from different streams, can be done simultaneously

Fix #624 
Fix #633

~To be merged after #654.~
closed 2023-12-23 by rafaellehmkuhl