bluerobotics/cockpit #634: Disallow widgets from going behind header/footer bars

ES-Alexander, 2023-12-11:
The header and footer bars (when visible) should be taken into account as invalid widget locations. It's quite disconcerting to resize the browser window and end up with some widgets partially covered by a bar.

This is the default map view, which doesn't seem reasonable:
[image]

The HUD has similar issues, with the compass at the bottom:
[image]

There _may_ be some leeway for full-screen widgets to be permitted to go behind them (it's kind of nice for the map, and I'm unsure whether it would be problematic for the video stream), but I think at minimum everything else should be confined to within the accessible screen area.

rafaellehmkuhl, 2023-12-11:
Yes. That's indeed one of the things we should be working between beta and stable. Thanks for opening the issue.

pkmiles, 2024-01-05:
My two cents on this - to keep it simple: 

- Header and Footer bars can be consistent across all views, i.e. no need to make the content on these view specific.
- View setting for whether it's full screen (under header / footer) or not.

Also, worth considering this because by doing this does it could simplify the view builder interface since the Header / Footer bar "viewlets" becomes separated...

rafaellehmkuhl, 2024-01-08:
> My two cents on this - to keep it simple:
> 
> * Header and Footer bars can be consistent across all views, i.e. no need to make the content on these view specific.
> * View setting for whether it's full screen (under header / footer) or not.
> 
> Also, by doing this does it simplify the view builder interface since the Header / Footer bar "viewlets" becomes separated? That could be a UX win.

I'm not totally sure about that. It seems to me that there's value in being able to create different bar layouts. Maybe we can give the user the option to use the same on all and have the best of both worlds?

PR #911 (2024-06-07): Move widgets away from bars
With this patch, widgets will not go under the top/bottom bar when the window is resized, nor the user will be able to move a widget to under the bars, with the exception being full-screened widgets.

I will keep this as draft for now to make sure I'm happy with the solution, and I've tested for all edge-cases.

Fix #634 

Before:

https://github.com/bluerobotics/cockpit/assets/6551040/9fca0261-f944-418a-82a2-60eef3b4d7d0

After:

https://github.com/bluerobotics/cockpit/assets/6551040/f7228c3c-023e-4fd9-a6df-e2b86a0dfe75

https://github.com/bluerobotics/cockpit/asse [... cut]
closed 2024-06-07 by ArturoManzoli