bluerobotics/cockpit #644: Joystick: allow axis protocol configuration

ES-Alexander, 2023-12-12:
Currently we forcibly send MAVLink `MANUAL_CONTROL` messages at a fixed rate, even though many vehicles don't support them, and some users may wish to change the frequency. It would be useful to allow turning that off and/or using some other protocol options for the motion control axes.

The most intuitive approach I can think of is providing a dropdown for the "axes protocol" which (for now) allows selecting between None, MAVLink `MANUAL_CONTROL`, and MAVLink `RC_CHANNELS_OVERRIDE`, together with a frequency selector/slider for the message rate (when the protocol is not None). That also leaves the UI open for additional protocols later, including non-MAVLink ones if that becomes relevant.

There is a bit of complexity involved though, because if the axes protocol is set to anything other than `MANUAL_CONTROL` then the corresponding button functions shouldn't be available to the user (since the messages won't be being sent). Perhaps we can allow them to stay selected (so they don't get unmapped), but just grey them out and/or strikethrough them to make it obvious they won't actually work.

rafaellehmkuhl, 2023-12-12:
Agree on most things.

And just to get one point correct, we only support `MANUAL_CONTROL` today, not `RC_CHANNELS_OVERRIDE`.

ES-Alexander, 2023-12-13:
> And just to get one point correct, we only support `MANUAL_CONTROL` today, not `RC_CHANNELS_OVERRIDE`.

Yes - I’m suggesting that `RC_CHANNELS_OVERRIDE` is a logical “next” protocol for us to support, which also illustrates why a dropdown would be useful rather than just a toggle switch between None and `MANUAL_CONTROL`.

rafaellehmkuhl, 2023-12-13:
> Yes - I’m suggesting that `RC_CHANNELS_OVERRIDE` is a logical “next” protocol for us to support, which also illustrates why a dropdown would be useful rather than just a toggle switch between None [...]

Got it. Seems reasonable.

ES-Alexander, 2025-08-21:
[QGC allows configuring the frequency, and selecting from various different output protocols](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/setup_view/joystick.html#advanced-settings).