bluerobotics/cockpit #653: Support trackpad regions as axis inputs/sliders

ES-Alexander, 2023-12-18:
No idea if this is possible (I'm not sure what the browser gets access to), but it'd be really cool to be able to assign regions of the trackpad as input sliders for controls, e.g. if I slide my finger up/down the right edge then I can make my lights1 dimmer/brighter, but if I do it on the left edge it does lights2 instead...

We could also then have the left and right halves of the trackpad used for motion control, which could be useful for testing purposes, or in emergencies if the joystick dies or something.

ES-Alexander, 2023-12-18:
May be possible using the [Touch events API](https://developer.mozilla.org/en-US/docs/Web/API/Touch_events), but from an initial read I'm unsure whether it provides access to touch positions relative to the pad, or just something based on previous mouse positions or something. The descriptions seem to be focused on what a touchscreen would have (where you literally touch parts of the page), but the API does specify it can apply to both touch screens and trackpads...

rafaellehmkuhl, 2023-12-18:
Oh you went far with that one 😅

I liked it, but apparently this is not yet supported by any browser api, although there is the [pointer lock api](https://developer.mozilla.org/en-US/docs/Web/API/Pointer_Lock_API), which is very cool and maybe has some synergy with that kind of feature.

ES-Alexander, 2024-03-30:
In case it's relevant, there's also the proposed [`GamepadTouch` Interface](https://w3c.github.io/gamepad/extensions.html#gamepadtouch-interface), but I don't believe it has browser support yet either.