bluerobotics/cockpit #656: We don't have a Cockpit Action for the most important feature: recording videos

rafaellehmkuhl, 2023-12-22:
We should add it, and in a way that the user can use it just with the joystick.

With the current `MiniVideoRecorder` behavior, the user would have to also click to select the desired stream to be recorded, but this adds 3 more clicks and the need to use the mouse/touchpad.

ES-Alexander, 2024-01-16:
Is this a duplicate of #610?

rafaellehmkuhl, 2024-02-07:
@ES-Alexander @pkmiles any ideas on how that should work?

I'm thinking of something like having a video recording action that is configurable.
- It could be used without being configured, in which case it would only work if there's only one stream available;
- If there's more than one stream available and it is not configured, it will trigger start/stop recording on all streams;
- The user will be able to associate the action to an specific stream, doing it on the video configuration page, or in an actions' configuration page;
- The user would be able to create many of those actions, one for each stream, to be able to control then individually;
- Alerts for start/stop of those would be emited so the user can have a feedback about what exactly is going on (e.g.: "Started recording stream 0", "Stopped recording stream 2", "Started recording all streams", etc);

What do you guys think?

rafaellehmkuhl, 2024-02-16:
Will close this one with #737 and open a new one referring to this, about adding the ability to record specific streams.

closed 2024-02-21 by rafaellehmkuhl