bluerobotics/cockpit #661: bug: If none of the allowed ICE IPs are available, no video is streamed

rafaellehmkuhl, 2023-12-23:


ES-Alexander, 2024-01-04:
Isn't that the expected behaviour? We could put up a warning saying "no allowed IPs are available for video streaming - please check the configuration" or something similar, but I don't think we should be creating/requesting a video stream from an IP address that the user has explicitly disallowed.

rafaellehmkuhl, 2024-01-04:
> Isn't that the expected behaviour? We could put up a warning saying "no allowed IPs are available for video streaming - please check the configuration" or something similar, but I don't think we should be creating/requesting a video stream from an IP address that the user has explicitly disallowed.

Agree.

PR #923 (2024-05-07): video: Warn user if they selected an IP that is not available in the ICE candidates
This situation leads to the ICE negotiation failing (since we are not allowing any of the available IPs), and thus no video stream being loaded.

For the future, we should change all those SweetAlerts by our own library, probably using the `InteractionDialog` component. This is probably part of #875.

Fix #661.
closed 2024-05-07 by rafaellehmkuhl