bluerobotics/cockpit #664: Inform user if the WebRTC stream negotiation failed

rafaellehmkuhl, 2023-12-23:
@Williangalvani suggested that this could improve the user experience, specially when making different integrations (as he's doing).

joaoantoniocardoso, 2023-12-23:
> @Williangalvani suggested that this could improve the user experience, specially when making different integrations (as he's doing).

Yeah, and it should keep trying the whole process from the beginning in case of failing.

Williangalvani, 2023-12-24:
For more information, I've seen three modes of failure:
 - ICE negotation fails for some reason (I suspect stun failure on MCM's side)
 - Lack of response on the signalling server, just no communication on the websocket
 - Available streams on the websocket but no ICE negoatiation

PR #890 (2024-04-19): Add feedback on video connection
This patch adds status for the signalling server and the stream connection on the video widget.
It also fixes some minors.

There's more to come, but I didn't want to hold bigger changes to get the fix in.

[image]

Fix #664 
Fix #684 
Fix #813
closed 2024-04-19 by rafaellehmkuhl