bluerobotics/cockpit #676: Camera Tilt mini-widget suggestion

pkmiles, 2024-01-05:
Might not even be a Cockpit thing... (vs. core software).  Is it possible to change camera tilt to:
- Show degrees not % - rough fixed calibration fine.
- Make 0deg facing horizontal.
- Suggest +ve then up and -ve then down.

ES-Alexander, 2024-01-08:
Agreed that this would be useful.

This could be handled entirely in the frontend (with an offset and a scaling factor), but it would also be good to use/support the autopilot parameters for vehicles that have them (`MNT_ANG(MIN/MAX)_TILT` for ArduPilot < 4.3, `MNTn_PITCH_(MIN/MAX)` for ArduPilot >= 4.3.0, not sure about PX4).* If we are using parameters (or supporting an offset) we'll need a dedicated mini-widget for it rather than the current default for the `VeryGenericIndicator`.

> *NOTE: For servo-type tilt mounts it would also be nice to support changing the servo PWM values (the servo used in the BlueROV2 can rotate to close to ±180° when the servo range is set appropriately beyond the 1100-1900µs default limits), but this might be out of scope for Cockpit right now, since BlueOS is currently in charge of peripheral setup and configuration

rafaellehmkuhl, 2024-01-08:
I'm curious as I don't have a servo-actuated camera: right now, 50% is the 0deg angle and 100% is upward max? Does it change from vehicle to vehicle?

As @ES-Alexander stated, for more detailed logic, like using the vehicle parameters, a dedicated widget would be needed, but I think this would be totally fine, as the idea of the `VeryGenericIndicator` is not to substitute everything, but just to help people see variables that do not have dedicated widgets yet or that just need to be showed on a generic way.