bluerobotics/cockpit #688: Record mini widget: Can we make persistent across multiple views?

pkmiles, 2024-01-18:
If you're on a view and decide to record something and then switch view it's unclear that recording continues because either:
- record widget not present on the view you've just switched to, or
- record widget is present, but doesn't show in recording state (with elapsed time).

BTW: I feel like this supports my previous comment about making header and footer common and persistent across all views....

rafaellehmkuhl, 2024-01-19:
Humn, good point Peter. I tend to agree with you on that.
I'm thinking we can allow the user to choose between having only one fixed bottom bar vs having one for each view.
What do you think about it? And should the one-per-view be the default or the opposite?
@ES-Alexander and @rjehangir I would also like to hear your opinion on that one.

joaoantoniocardoso, 2024-01-19:
@rafaellehmkuhl, allow me to refresh us on some of what was previously mentioned in some of our meetings, as it might be useful now:

---

### A) Tag system
Following DWM, we could say:
>    _Widgets are grouped by tags. Each widget can be tagged with one or multiple tags. Selecting certain tags displays all windows with these tags._

The UX could be as follows: To add/remove, the user could right-click on the widget, and this would open a menu in which a submenu is 'Tags', which will have a couple of checkboxes for each view plus the options 'all' and 'only current'.

### B) Widget container/bar

A new widget, that acts like a bar, where you can add multiple widgets, and being a widget itself, it could be added to (but not limited to) the bottom or top bar. Its configuration would be global, but the user would still have to add it to multiple views.

### C) Pinned widgets

A 'pin' button/configuration to make a widget global, making it be shown on all views.

### D) Pin bar

A 'pin' button/configuration to make the bar global.

---

Note that they are not necessarily conflicting options.

rafaellehmkuhl, 2024-01-19:
> @rafaellehmkuhl, allow me to refresh us on some of what was previously mentioned in some of our meetings, as it might be useful now:
> 
> ### A) Tag system
> Following DWM, we could say:
> 
> > _Widgets are grouped by tags. Each widget can be tagged with one or multiple tags. Selecting certain tags displays all windows with these tags._
> 
> The UX could be as follows: To add/remove, the user could right-click on the widget, and this would open a menu in which a submenu is 'Tags', which will have a checkbox for each view plus the options 'all' and 'only current'.
> 
> ### B) Widget container/bar
> A new widget, that acts like a bar, where you can add multiple widgets, and being a widget itself, it could be added to (but not limited to) the bottom or top bar. Its configuration would be global, but the user would still have to add it to multiple views.
> 
> ### C) Pinned widgets
> A 'pin' button/configuration to make a widget global, making it be shown on all views.
> 
> ### D) Pin bar
> A 'pin' button/configuration to make the bar global.
> 
> Note that they are not necessarily conflicting options.

- The tag system I personally don't like. It's complex, confuse, and in terms of UX seems like the worst to me.
- The widget container is what we have right now, although we don't have the option today to use the same container on multiple bars, which sounds like a good solution for the feature (both re-use or copy/duplicate it, so the user can change the former).
- The pined widgets sounds good initially, although I have questions on the behavior, for example: where should the widget be positioned if there are other widgets around?
- Pinning one bar seems like an easy and intuitive solution.

ES-Alexander, 2024-01-26:
I think that 
- too much configuration is likely confusing, but 
- different Views could benefit from showing different telemetry
    - because they tend to represent different use-cases / operating modes

I would personally lean towards not allowing the user to choose which bars are pinned, and instead 
- always pin/share the top bar and part of the bottom bar across Views, and then 
- the middle (and possibly one of the sides?) of the bottom bar can be clearly documented and indicated as variable between Views.

---

Pinning and/or linking arbitrary widgets across views (including mini-widget containers) could be _fun_, but also seems like more complexity than it's worth, especially if we
1. make it easy to duplicate widgets across views
    - some UI suggestions for that [here](https://github.com/bluerobotics/cockpit/issues/699#issuecomment-1912078827)
4. make multiple widgets able to efficiently draw from the same data source
    - e.g. for video widgets and recording widgets