bluerobotics/cockpit #696: Views: Include enable/disable in settings

pkmiles, 2024-01-22:
Sometimes it's useful to disable certain views for display, but not remove them entirely from the view configuration.  Would it be possible to include a Display Enable yes/no checkbox in the Settings for each View?

pkmiles, 2024-01-22:
Use case example: We don't have GPS integration so would like to disable Map View, but we might add it soon, so don't want to have to remove it and then build from scratch...

I guess an alternative (which may already be supported - would need to check) is to be able to copy a View from one Config to another.

rafaellehmkuhl, 2024-01-23:
It is possible and definetly desirable!
I will include this issue in our Beta->Stable milestone.
It's an easy one, so I will probably tackle it during the next week.

PR #708 (2024-01-26): Allow toggling visibility of Views
[image]

@pkmiles you might want to check that out :)

Fix #696
closed 2024-01-26 by rafaellehmkuhl