bluerobotics/cockpit #697: Altitude widget (left hand side of HUD screen perhaps?) - vs. Depth widget

pkmiles, 2024-01-22:
It would be useful to have an altitude widget, similar design to the depth widget (except positive is up), that could be put on the left hand side of the HUD.  There may be plans to show altitude in addition to depth...

rafaellehmkuhl, 2024-01-23:
Agree.
We did an altitude indicator (mini-widget) but didn't add a regular widget version.
The original plan was to have the depth widget more flexible, so it could handle both cases, but it's easier for us to start with a separated one.

ericjohnson97, 2024-01-23:
@pkmiles Like @rafaellehmkuhl said there is a mini widget for relative altitude. 

One possible to way to make it show on the left hand side would be to add a "Mini Widget Bar" which is a regular widget and can be placed anywhere on the screen then add the relative altitude mini widget to it. 

If you would like to implement a full size widget you can look at https://github.com/bluerobotics/cockpit/blob/master/src/components/mini-widgets/RelativeAltitudeIndicator.vue for reference

[image]

ES-Alexander, 2024-01-26:
I think a good altitude indicator should be included within the depth indicator, especially since the features are so similar, and it leaves scope for advanced vehicle types / operating conditions in future.

In case it's of interest, there are some UI ideas in #247 - [this comment](https://github.com/bluerobotics/cockpit/issues/247#issuecomment-1803039907) was specifically about aerial use-cases.