bluerobotics/cockpit #699: View edit UX design: Move top level vehicle/config choice to settings to provide more space...

pkmiles, 2024-01-22:
Just a thought: The View Edit sidebar is pretty crammed. The reality is people are not going to be flicking from ROV to Boat to some other device. How about moving that top level (and associated functionality) across to Settings. This would then allow you to have more realestate for view selector (which could be a drop down list to save more space) and scrolling list of widgets.

I realise this stuff is probably a lot of work and these designs are mature - but thought I'd raise it!

ES-Alexander, 2024-01-26:
I was just about to make a related issue, and came across this one, so I'll add my thoughts here.

I think it's ok to keep the profiles in the edit menu, but I think we should change the display to be a nested tree with expanding sections - something like this:
[image]

Benefits to this approach include:
1. the tree/nesting depth intuitively matches the relationship between Profiles/Views/Widgets
    - i.e. the lower level elements are shown "within" the component/structure that contains them
1. the section you're actually working in takes up most of the display area
    - the other sections can be above/below, and scrolled away from as relevant
1. there are fewer control buttons shown at once, which helps to focus on the actual components
    - to further reduce clutter, we may want to consider only showing the configuration buttons while the mouse is hovered over the component, although I'm not sure if that could cause issues with touch-screen users

And as some advanced interaction options:
1. views (and profiles?) could potentially be intuitively dragged and reordered within the tree, in the same way widgets can be currently
1. widgets could be dragged between views, and/or duplicated to a different view (e.g. dragging while holding a modifier key/button)
1. views could be dragged between profiles in the same way
1. the icons and names of components could be editable by just clicking on them directly

rafaellehmkuhl, 2024-01-26:
Wow. This actually looks very good!

pkmiles, 2024-01-26:
Yep. awesome.

rafaellehmkuhl, 2024-07-02:
@ArturoManzoli this one was marked as Done by mistake, right?

ArturoManzoli, 2024-07-02:
> @ArturoManzoli this one was marked as Done by mistake, right?

Yes, it was a mistake. Thanks for reopening

closed 2024-07-22 by ArturoManzoli