bluerobotics/cockpit #714: Joystick: allow multiple inputs to control the same axis

ES-Alexander, 2024-01-31:
The current behaviour is for only one input to be able to control a given axis, but as raised [here](https://github.com/bluerobotics/cockpit/pull/711#pullrequestreview-1852922970) there are benefits to supporting multiple inputs assigned to the same axis, for various advanced control scenarios.

In particular,
> with "conflict" resolution handled by just ***summing up the inputs from all inputs mapped to a given axis***:
>
> - as Willian has previously suggested this would allow using the triggers as up/down inputs
> - it would also allow using something like a slider to set a general level for an axis, then a spring-loaded joystick to handle finer-grain control around that set point
> - it could be used as a power limiter, whereby joystick 1 can only do up to 50% of the possible axis range, and consciously also using joystick 2 allows "overloading" to use higher levels, which some controllers might want to use as a safety mechanism (in the same way that you need two keys to unlock some doors, like if the vehicle could restart itself if you overload without knowing what you're doing)
> - it could also set things up nicely for multi-joystick control, with a trainer and student, especially if we could set some kind of ***numerical weighting to the values from each input*** (rather than requiring one to be a strict override of the other)

ES-Alexander, 2025-05-15:
This could be effectively accomplished using a Compound variable with #1658, inside the widget configuration that would be made possible by #1759.

rafaellehmkuhl, 2025-05-16:
Yeah. And it's basically what we are doing for the camera zoom and focus controls already.

> This could be effectively accomplished using a compound data-lake variable with [#1658](https://github.com/bluerobotics/cockpit/issues/1658), inside the widget configuration that would be made possible by [#1759](https://github.com/bluerobotics/cockpit/issues/1759).