bluerobotics/cockpit #715: Improve video download pipeline to guarantee (as much as possible) that recorded videos are not lost

rafaellehmkuhl, 2024-01-31:
We had [a recent report from Peter](https://discuss.bluerobotics.com/t/cockpit-video-stream-recording-functionality-and-problems-with-download/15922) that he did some video recording and the download failed, resulting in him losing his footage.

This is probably one of the worst possible situations for the user, as you can't simply film something again.

Cockpit should do whatever is in its hands to prevent this kind of problem.

The first thing that comes to my mind is simply not auto-deleting video recordings. Not even if we had the download confirmed (we can't know if it had actually finished with success). We should always stay with the footage in the disk, and let the user know that there are footage there that can be deleted if necessary.

pkmiles, 2024-02-01:
Thanks @rafaellehmkuhl . Looks like a good solution. You could even have a "space remaining" mini widget - akin to how GoPro shows you how much is left on your SD card.   

Aside: I posted that on the forum because I wasn't sure at that point of whether to raise it as an issue, plus I thought it might be useful for the wider user community to understand the core capability of Cockpit regarding video recording.  Let me know if that's a problem - I can understand you may not want to expose the wider group to the sausage making that goes on while products in beta.

rafaellehmkuhl, 2024-02-01:
Hey @pkmiles! No problem posting on the forum, really. It's indeed good that we have this kind of discussion there (your points on what's the goal with the recording functionality specially). Feel free to open the topics there or the issues directly here. I will do my best to organize them afterwards. We are also not sure yet on what exactly should go on the Discuss vs Discord vs Github 😄

pkmiles, 2024-02-01:
Another thought. Maybe the way video management could work is, rather than downloading after you stop recording, storing them on the ROV (with the memory mini widget so you know when you're nearing full) and then requiring a specific action to download. After download complete (and ideally validated) the user then has the option to remove the videos from storage.

This experience would be akin to pulling an SD Card out of GoPro, transferring the videos, ensuring they're all there and then deleting them off the card (or reformatting the card).

Thoughts?

rafaellehmkuhl, 2024-02-02:
About the "record and then decide what to do" experience, that's actually exactly what I was proposing yesterday in our internal meeting 😃 

To start, we would keep the video being stored on the browser DB (a.k.a. your computer's disk), as it's not yet well-defined in BlueOS how the extensions should hold temporary files, but as soon as we decide that for BlueOS, I think we should move the videos also there (as it's being done in #701 for Cockpit's settings).

PR #720 (2024-02-07): Stop saving video on memory and stop removing videos automatically
https://github.com/bluerobotics/cockpit/assets/6551040/6a1c6904-eea7-41a8-b094-8c424ef076b8

Fix #645 
Fix #693 
Fix #715 

Notice, *I think* this still does not completely solve the problem of recording large video files. Although we don't use the RAM to store the video anymore, RAM is still used when you click to download the video file, as we need to put the file in memory to start the download.

@pkmiles once we merge this PR, could you help testing the new Beta with it? It would be specially cool to test it with the 4K cam [... cut]
closed 2024-02-07 by patrickelectric