bluerobotics/cockpit #719: Store video recording on the vehicle instead of the top-side computer

rafaellehmkuhl, 2024-02-02:
The same move we are doing for the settings in #701, we should do for the video recordings.

rafaellehmkuhl, 2024-02-02:
> this should probably be linked to an issue in Blueos Or MCM

That would probably be [this issue](https://github.com/bluerobotics/BlueOS/issues/1640) in BlueOS.

About MCM, I'm not sure if we want to use it for that.

@joaoantoniocardoso @patrickelectric what do you guys think?

joaoantoniocardoso, 2024-02-02:
Sorry, can you explain what is the relationship to MCM here? I'm lost

Williangalvani, 2024-02-02:
> Sorry, can you explain what is the relationship to MCM here? I'm lost

Using it we could save video directly to the disk instead of having another service in the rov talking to it in order to store the video streams.

wait. is this issue originally considering upload the video back to the vehicle? that sounds unproductive

rafaellehmkuhl, 2024-02-02:
> wait. is this issue originally considering upload the video back to the vehicle? that sounds counter-productive

Yes it is. As we don't have an interface for it today in MCM, that would be my approach, unless we can get this functionality there.

joaoantoniocardoso, 2024-02-02:
hahahah yeah, it's easier to implement the video-saving feature in MCM instead.

rafaellehmkuhl, 2024-02-02:
> hahahah yeah, it's easier to implement the video-saving feature in MCM instead.

Cool. Let's discuss that next monday.