bluerobotics/cockpit #726: Joystick: Allow use axis as button

RaulTrombin, 2024-02-08:
Issue:
can't map buttons using d-pad (axis 6 & 7);
[image]

Tryng to set 8BitDO controller;
Details:
OS: 
Linux ThinkPad-T495 6.5.0-17-generic #17~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan 16 14:32:32 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
Controller:
8Bitdo SN30 Pro (Vendor: 2dc8 Product: 6101)
*power on D-mode ( B + Start ), pair with Ubuntu as normal.
*X-mode doesn't work properly

ES-Alexander, 2024-04-08:
The simplest approach I can think of for this would be allowing each axis to be configured as a function that triggers when the axis moves above or below a given threshold.

More advanced ideas could allow:
- supporting some form of automated hysteresis/debouncing
- user-selectable length of time in a particular value region before triggering the associated function
- assigning a variable-length list of functions to specific value ranges
    - e.g. for multi-position sliders and knobs that aren't spring-loaded
- passing the axis value directly to some kind of Action
    - relevant to #644, but focused more on general functionalities rather than specifically motion control axes
    - likely with some kind of scaling and offset options as "calibration"
        - relevant to #616
    - would be helpful for making full use of #826
    - this should maybe be its own issue, but raising it here since the features are related

ES-Alexander, 2025-07-16:
This will be _slightly_ less complex with compound variable support - an axis could be assigned to a variable, then that variable is part of a compound variable with the threshold in it, and the compound variable is then used to trigger an Action.

That said, that workflow would still be very much a workaround, rather than being nicely configurable from in the joystick interface. Ideally there'd be a nice interface which automatically creates and manages the relevant direct and compound variable in the background (related to #1883).