bluerobotics/cockpit #727: Widgets: Idea: Add a screen-based joypad for pan/tilt camera

RaulTrombin, 2024-02-08:
Something like:
https://bobboteck.github.io/joy/joy.html

But also with a slider bar to control zoom and focus;
[image]

rafaellehmkuhl, 2024-02-08:
This would be valuable for overall vehicle control also!

We should have that, yes. For that we need to change our structure a little bit to be able to accept inputs for things other than the Gamepad API.

ES-Alexander, 2024-09-23:
Potentially relevant to #1291

ES-Alexander, 2026-08-31:
Since #2476 has been merged, the 'backend' functionality of this can be plumbed into the normal joystick control pipeline, as long as the widget provides configuration for choosing a couple of data lake variables to control.