bluerobotics/cockpit #730: we shouldn't allow configuring the joystick when armed.

Williangalvani, 2024-02-08:
we should either block or automatically disarm. I'm not sure which one is better.

rafaellehmkuhl, 2024-02-09:
Agree.

@bluerobotics/software-team any thoughts on that?

ES-Alexander, 2024-02-09:
My preference would be for the whole burger menu to be disabled while the vehicle is armed, and the only MAVLink messages that should be sent to a vehicle while outside the fly and mission planning views (i.e. while in a configuration page, or Cockpit is out of focus) should be heartbeats (if they’re enabled).

rafaellehmkuhl, 2024-02-15:
> My preference would be for the whole burger menu to be disabled while the vehicle is armed, and the only MAVLink messages that should be sent to a vehicle while outside the fly and mission planning [...]

Yeah, that's more aggressive, but maybe right. Can you think of any situations where the user needs to be able to go to the configuration or edit menu while armed?

Williangalvani, 2024-02-15:
> Can you think of any situations where the user needs to be able to go to the configuration or edit menu while armed?

I can think of one. If there is a water current and the ROV is in position hold, it would be safer to keep it armed than to disarm it before you open the configuration menu

Williangalvani, 2024-02-15:
I now think we should just have a confirmation dialog "Your vehicle is currently Armed. Disarming before opening the configuration is advised." or similar, allowing either to disarm or dismiss it

PR #784 (2024-02-29): Add check before opening hamburger config menu while vehicle is armed
Add a check before opening hamburger configuration menu while vehicle is armed to avoid users changing configurations while vehicle still armed preventing potential dangerous actions
Closes #730
closed 2024-02-29 by rafaellehmkuhl