bluerobotics/cockpit #731: iFrame widget content reloads when changing views

rafaellehmkuhl, 2024-02-09:
Reported by @rjehangir:

> It would be cool if the iframe could maintain a session even if you change the view and come back. Right now it reloads the page every time.

JoaoMario109, 2024-06-18:
The issue with the iFrame reloading, even though the component is not "unmounted" in Vue, is probably related to how widgets are rendered on the screen in this [code](https://github.com/bluerobotics/cockpit/blob/a1111da5374038f17dedc951dd47f9b78fa93900/src/views/WidgetsView.vue#L3).
This results in the widget being either removed or moved in the real browser DOM. According to information in these links:

* [Microsoft Documentation](https://learn.microsoft.com/en-us/previous-versions/windows/internet-explorer/ie-developer/compatibility/gg622929(v=vs.85)?redirectedfrom=MSDN)
* [Stack Overflow Discussion](https://stackoverflow.com/questions/8318264/how-to-move-an-iframe-in-the-dom-without-losing-its-state)

So it seems that it is impossible to move or remove an iFrame in the real browser DOM without it being freed and recreated, causing the unwanted reload. A simple test for this is to set all lifecycle hooks in Vue and print debug messages for these events. Use a YouTube video in an iFrame, and when switching views, observe that the hooks are not called, but the video stops and reloads when returning to the original view containing the iFrame. This implies that the iFrame is being removed and freed under the hood in the real browser DOM.

So to address this  we will likely need to handle iFrame widgets specially by ensuring they reside in a separate parent element that never removes or moves them, and we just control their visibility and positions.

rafaellehmkuhl, 2024-06-20:
More on that on the [whatwg repo](https://github.com/whatwg/html/issues/5484).

PR #1022 (2024-06-27): iframe: Persist position on DOM so it doesn't reload on view change
Right now I'm positioning the actual iFrame element. 

~I will try to do that with the iFrame widget, so it doesn't add extra position/size logic and supports native widget features (like the toolbar repositioning).~ I think we can leave with that for now, so I've opened #1043 to track future improvements on that.

https://github.com/bluerobotics/cockpit/assets/6551040/56d144df-721c-4320-915a-08cb6ea140ea

Fix #731
closed 2024-06-27 by patrickelectric