bluerobotics/cockpit #736: "Clear temporary video cache" button does not work properly if there are saved videos

rafaellehmkuhl, 2024-02-16:


pkmiles, 2024-02-16:
hi @rafaellehmkuhl . I found this button a bit confusing to be honest. 

The .webm and .ass files you see above this button: These are whats stored on the Raspberry Pi SSD, right? So on top of that is there a temp video cache before they get written to the SSD (?) and that button clears it(?).  

If above statement TRUE:
1) When would a user need to do this? i.e. is it a routine activity or something you'd do when something goes wrong/gets stuck. 
2) If so i'd say that's kind of a nuts-and-bolts exception feature and should be in "advanced" and hidden on the page with a bit more of a description on when and why you should use it.

If above statement FALSE (and Temp Video Cache = SSD storage of .webm and .ass files):
1) Then maybe look at simplifying the language around this: "Device Storage" or "Local Storage" something like that? 

I think as soon as you start using the word "cache" in a user interface you're potentially losing a non-technical audience (lol) What do you think?

rafaellehmkuhl, 2024-02-19:
Answering in parts:

0. Both the .webm and .ass files are stores on the disk of your topside computer. We currently don't use the vehicle storage for that, although we plan to (#719). The difference between the .webm/.ass files available there and the temporary files (mentioned by the button) is that the former are final files to be read by a video player, while the latter are video chunks of about 1 second that are stored during recording and merged on a single file when you click to stop the recording.

1. The user should use this button regularly, to prevent unwanted usage of disk space, and this is not done automatically today so we can try to recover from failed videos when they happen during this beta phase. We still have to decide if we are going to keep it there or just do that automatically once we reach stable.
2. Totally agree.

So, in the end, we still have to decide on what to do with this button, but your suggestion of putting it on an advanced mode is good.

We also have to add the option for the user to download those temporary files for the cases of video recovery, like you reported on #741. I commented about it more there.

PR #752 (2024-02-21): Improve video download page more
[image]

Fix #736
Fix #751 
To be merged after #750
closed 2024-02-21 by rafaellehmkuhl