bluerobotics/cockpit #741: Video recording: Issues with shutting recording off, and videos captured found to be corrupted on download

pkmiles, 2024-02-16:
Field trip yesterday.  I'm on Beta 7 at the moment. 

Some weird experiences worth digging into: 
1) Clicking record successfully started video, but when attempting to stop the video it wouldn't work.  I had to change Views in order to stop recording.   I had a quick go at bench testing this morning, but failed to recreate this... Will spend more time later. Only difference I can see is that we're on a network / internet here. No network / internet on the boat.  Watch this space.

2) The videos were showing on the downloads page yesterday, so I assume they'd failed. Then today, when I bench tested, they were there...  I downloaded them... But they appear to be corrupted. Hefty files, so something there... Just won't open in a browser (Chrome) or VLC.  I've included a video from a pool test we did friday - which did work.  No internet / wifi there either. (salt water vs. fresh water ? lol). 

I'll keep digging in on trying to understand these problems, but let me know if you want anything specific tested. Sorry, I still haven't managed to do the 4k camera duration recording test. On my list!

Here's the videos:
https://www.dropbox.com/s/dgih1l4s3d531lq/Videos%20from%20Peter.zip?dl=0

(>Github's 10MB limit)

rafaellehmkuhl, 2024-02-19:
Also, @pkmiles you still have your "temporary files" right? We can try to retrieve the video chunks from there and analyze it to better understand what happened, and maybe even fixing the videos.

joaoantoniocardoso, 2024-02-19:
Hi!

For investigating video-related things, usually [MPV](https://mpv.io/) helps more than VLC, especially when in verbose mode (`mpv -v <videofile>`).

When playing the biggest video file on MPV, it looks like multiple clips are being mixed into the same file.

rafaellehmkuhl, 2024-02-19:
> Hi!
> 
> For investigating video-related things, usually [MPV](https://mpv.io/) helps more than VLC, especially when in verbose mode (`mpv -v <videofile>`).
> 
> When playing the biggest video file on MPV, it looks like multiple clips are being mixed into the same file.

Pefect joao!
Tried here with MPV, and it worked, although the clips do not have the duration fixed and indeed seems like more than one footage merged (or maybe a very hard package loss?).

PR #807 (2024-03-06): video: Stop recording counter as soon as user clicks to stop
Fix #741 
Fix #766
closed 2024-03-06 by rafaellehmkuhl