bluerobotics/cockpit #748: src/stores/video.ts: review

joaoantoniocardoso, 2024-02-20:
A brief review of `src/stores/video.ts` with the problems reported on #741.

### 1) src/stores/video.ts:156:
```typescript
    const fileName = `${missionStore.missionName || 'Cockpit'} (${format(
      streamData.timeRecordingStart!,
      'LLL dd, yyyy - HH꞉mm꞉ss O'
    )})`
```

The database key is not guaranteed to be unique: two recordings starting simultaneously (here, at the same second) will create conflicting database entries.

### 2) src/stores/video.ts:166:
```typescript
    activeStreams.value[streamName]!.mediaRecorder!.ondataavailable = async (e) => {
      await tempVideoChunksDB.setItem(`${fileName}_${chunksCount}`, e.data)
      chunksCount++
    }
```

we can mistakenly overwrite previously stored chunks in the database if we have a conflicting entry.

### 3) src/stores/video.ts:187:
```typescript
      const sortedChunks = chunks
        .sort((a, b) => {
          const splitA = a.name.split('_')
          const splitB = b.name.split('_')
          return Number(splitA[splitA.length - 1]) - Number(splitB[splitB.length - 1])
        })
        .map((chunk) => chunk.blob)
```

We are unsafely using `missionName` as part of the key for the chunks in the database, allowing the user to break the naming scheme parsing logic (e.g. by simply introducing underscores (`_`) in the name)

rafaellehmkuhl, 2024-02-20:
Agree with all of them. We should work on that for stable.

rafaellehmkuhl, 2024-02-21:
Actually number 3 would never happen, because we are taking the last member of the split array, which will always be our chunk number, but I will change it as well.

joaoantoniocardoso, 2024-02-22:
Sorry, but (3) is still a problem. e.g.:
[image]

joaoantoniocardoso, 2024-02-22:
My bad. I thought it was still using the stream name still!

But now we might have another problem. When we are doing this:

```typescript
        const sortedChunks = chunks
        .sort((a, b) => {
          const splitA = a.name.split('_')
          const splitB = b.name.split('_')
          return Number(splitA[splitA.length - 1]) - Number(splitB[splitB.length - 1])
        })
```

My mental compiler might be wrong today, but I think we are having a `NaN - Number` here, because:

```typescript
chunkName = `${recordingHash}_${chunksCount}`
```
and
```typescript
const recordingHash = uuid().slice(0, 6)
```

rafaellehmkuhl, 2024-02-22:
Didn't get it. 

The chunk names will be something like `abcdef_0`, `abcdef_1`, and so on.
We will be accessing ['abcdef', 0][1].

joaoantoniocardoso, 2024-02-22:
thanks, now it made sense, sorry!

closed 2024-02-22 by joaoantoniocardoso

rafaellehmkuhl, 2024-02-22:
No problem haha

Thanks for the review on the code logic. It was problematic indeed.

PR #756 (2024-02-22): Improve video chunks handling
Fix #748