bluerobotics/cockpit #777: If a recording is ongoing and the user closes the notebook lid, the video is stopped but not saved

rafaellehmkuhl, 2024-02-27:
Reported by @vshie on Beta9.

Williangalvani, 2024-02-27:
I wonder if that is why I lost footage from my last test 🤔

rafaellehmkuhl, 2024-02-28:
The problem is actually more generic. It does not create the video from the chunks when the tab is reloaded with a recording ongoing.

I tested here, and by closing the lid, waiting some time, then opening again, the browser asks me if I indeed want to close the tab. If I click to cancel, Cockpit stops the recording and creates the video file. If I click reload, it does not create the video file. This also happens when one reloads the tab (without closing the lid).

I'm changing the title to better reflect that.

PR #791 (2024-03-05): Allow recovering unprocessed videos
This patch gets back the functionality of recovering videos when Cockpit was interrupted during a recording (tab closing, shut down, lid closed, etc).

On top of that, it also allows for a recovery on situations where for some reason a processing (triggered by the stop button) failed.

To be merged after #788.
Fix #777.
closed 2024-03-05 by rafaellehmkuhl