bluerobotics/cockpit #786: Support `SURFTRAK` mode for ArduSub

ES-Alexander, 2024-02-28:
### Description

`SURFTRAK` surface tracking / range holding was added in ArduPilot/ardupilot#23435, and will be released in ArduSub-4.5. It would be nice if Cockpit had explicit support for it, especially given the poor user experience highlighted by #436.

### Additional Context
- Mentioned in [this forum comment](https://discuss.bluerobotics.com/t/rov-mass-buoyancy-distribution-and-compensation/9877/15)
- [QGC PR](https://github.com/mavlink/qgroundcontrol/pull/11126) in case it's relevant for reference

closed 2024-02-29 by rafaellehmkuhl

clydemcqueen, 2024-06-04:
Also: it would be helpful if the [generic indicator](https://github.com/bluerobotics/cockpit/blob/master/src/types/genericIndicator.ts) supported the [new RFTarget variable](https://github.com/ArduPilot/ardupilot/blob/bd9102212074e680f781dcfaebee7a1b0989c5c6/ArduSub/GCS_Mavlink.cpp#L148).

rafaellehmkuhl, 2024-06-04:
> Also: it would be helpful if the [generic indicator](https://github.com/bluerobotics/cockpit/blob/master/src/types/genericIndicator.ts) supported the [new RFTarget variable](https://github.com/ArduPilot/ardupilot/blob/bd9102212074e680f781dcfaebee7a1b0989c5c6/ArduSub/GCS_Mavlink.cpp#L148).

Actually the VGI supports all MAVLink variables by default. If the variable is in the messages, it will appear there to be selected.

PR #787 (2024-02-29): Add `SURFTRAK` mode support for ArduSub
Fixes #786.

⚠️ **UNTESTED** - I just figured I'd get the code started, and hopefully someone can comment if there's something obvious I'm missing.

If possible it would be nice to have the mode and joystick button disabled for firmwares that don't support them (e.g. for non-ArduSub vehicles, and ArduSub <4.5), or at least provide an obvious warning when trying to switch to it (this may already happen if the vehicle rejects the mode change - not sure), but I'm not sure how to do either of those things.

@clydemcqueen may have some input as well.

---

@r [... cut]