bluerobotics/cockpit #792: Cockpit should store map tiles so users that boot offline can use the map features

rafaellehmkuhl, 2024-02-29:
Right now they have to open Cockpit while online and never refresh.

ES-Alexander, 2024-03-05:
My preferred approach for this is if it's handled in BlueOS / a separate extension/add-on (per bluerobotics/BlueOS#1820), so extensions don't each need to download their own map data.

If that's managed well the maps service could handle relevant APIs for downloading from different mapping sources, and any extension could use the service's API to request whatever map area + source it wants.

An associated add-on could provide a link to the folder where the current maps data is stored for any extensions that want it, so they only need to request data if what they want isn't already there, and if they support adding custom maps/overlays then they can add them to the folder directly to allow other extensions to make use of them if they want to.

Williangalvani, 2024-12-19:
this ins interesting. paid for commercial use, but interesting

https://data.maptiler.com/downloads/planet/

Williangalvani, 2024-12-19:
other relevant links:
https://wiki.openstreetmap.org/wiki/Tile_servers

rafaellehmkuhl, 2024-12-19:
I don't think we should rely on BlueOS for such a critical feature on the long term (although I'm not against someone implementing it over there for an initial version).

Williangalvani, 2024-12-19:
Yeah I'm actually thinking of how to handle user-generated tile data. Like
drone shots, photogrammetry stuff, sonar mosaics, and so on. For smaller
areas we could just use cesium's single tile entry with the full thing, but
I feel like that wouldnt scale

Williangalvani, 2025-01-09:
@pkmiles cool idea! we could get the camera fov and scale the image size based on rangefinder+rov depth, too.

ES-Alexander, 2025-08-21:
[A maritime chart format](https://en.wikipedia.org/wiki/S-63_(encryption_standard)), [raised in the forum](https://discuss.bluerobotics.com/t/how-to-integrate-s-63-encrypted-charts-enc-into-qgroundcontrol/21077))

rafaellehmkuhl, 2025-08-21:
> [A maritime chart format](https://en.wikipedia.org/wiki/S-63_(encryption_standard)), [raised in the forum](https://discuss.bluerobotics.com/t/how-to-integrate-s-63-encrypted-charts-enc-into-qgroundcontrol/21077)

Just for us not to miss it, I think the first goal of this functionality is the ability to simply cache the current tiles for offline usage, and use them when offline, without anything extra. Today the lack of this feature effectively impedes offline missions.

Adding extras, like allowing other sources, or even custom sources, should be on a separate issue.

PR #2085 (2025-09-29): Add option to store map tiles for offline missions
* Added option to download all map tiles on the current displayed area;
* Downloads include tiles from the current zoom level up to the maximum zoom level.
* Added a download status indicator (some downloads can take a while, specially on far zoom levels);
* Tiles are shared between Mission Planning and the Map widget;

https://github.com/user-attachments/assets/decf74e9-1f01-414d-a2f3-0d0e696f49c1

Closes #792
closed 2025-09-29 by ArturoManzoli