bluerobotics/cockpit #801: Improve video debugging support

ES-Alexander, 2024-03-04:
Expanding on some suggestions in #800, we should support:

- [ ] an optional debug/quality overlay in the Video widget, showing FPS, bitrate, and dropped frames
- [ ] optional recording/logging of those values
    - relevant to #621 if logging to the subtitle files
    - even if it's just stored in Cockpit's logs and saved/accessed separately, that should still be much faster to check than scrubbing through a long video recording
        - if we do this approach then we should probably run an automated video quality check after each recording, and notify of potential problem periods and/or provide a graph of video health over time
- [ ] a "video health" mini-widget, tied to an incoming video stream in the same way the recording widget is
    - I'm imagining a video symbol (per connected stream) in the top right mini-widget bar, with the symbol colour/icon based on thresholds for quality, e.g.
        - green video-camera --> all good
        - yellow video-camera --> some dropped frames, or inconsistent FPS/bitrate
        - yellow video-camera with black exclamation --> bad enough that investigation is recommended
        - red video-camera with black exclamation --> poor quality - footage unlikely to be usable
        - grey video-camera with red cross --> camera stream not detected as available
    - hovering over the icon should show status highlights (i.e. the cause of the status, like "low FPS, 21.3/25"), and any corresponding recommendations (check power supply, camera temperature, etc)
    - clicking on the icon should either 
        - open a pop-up showing the video quality info + common troubleshooting tips, or 
        - enable the debug overlay for any corresponding video widgets that are using that stream

Williangalvani, 2024-03-14:
relevant code:
https://github.com/selkies-project/selkies-gstreamer/blob/main/addons/gst-web/src/app.js#L387-L434

ES-Alexander, 2025-07-16:
#1179 and #1403 implemented the first two ideas.