bluerobotics/cockpit #811: Video recorder: indicate download availability

ES-Alexander, 2024-03-07:
Currently when videos are recorded there's no indication of when they're finished processing (and ready for download), or how many are available.

Expanding on some discussion from #807, it would be nice if each video recorder mini-widget keeps track of the videos it has recorded, insofar as it displays a loading/waiting icon while a video is processing, and a download icon when there's a video available for download (ideally with a counter showing how many are available). Clicking on the icon could either take the user to the videos page, or could just directly download the video(s).

I think this icon should be an extra UI element, that's part of the widget but not part of the main recording button (so that new recordings can still be started while one is processing / available for download).

Example UI ideas:
[image]

rafaellehmkuhl, 2024-03-07:
Would this separate UI element be part of the widget? In this case, the idea is that when this element appears, the recorder widget uses more space in the bar (pushing the other widgets)?

ES-Alexander, 2024-03-07:
Yes it would be part of the widget (would be nice if multiple recording widgets could track only the recordings of the video streams they’re linked to).

I'm not sure whether it makes more sense to add an extra bit on the end when an icon should be shown (thus moving the other things in the bar), or if the widget should always take up that room and then the main button just shrinks when one of the icons is made visible.

rafaellehmkuhl, 2024-03-07:
I really liked that second approach (increasing the default widget width and changing what we do with it on the fly).

ES-Alexander, 2024-03-14:
@ArturoManzoli it looks like you've basically fixed this in #821, but with a slightly different approach to the "processing" state handling.

Ideally we'd have the processing indicator as part of / in addition to the available downloads, and return the recording button immediately to its idle state so that it's obvious to users that they can start a new recording even while the previous one is processing.

ArturoManzoli, 2024-03-15:
@ES-Alexander I agree with you. The processing indicator as part of the available downloads is a batter approach and will be implemented soon. But before that we have to analyze if its alright to start a new recording while processing the previous one. An issue #823 has been created to investigate that.

closed 2024-03-21 by ArturoManzoli