bluerobotics/cockpit #812: When reconnecting to a WebRTC stream, 2 streams are requested.

joaoantoniocardoso, 2024-03-07:
How to reproduce

1. turn on the vehicle w/ mavlink-camera-manager running
2. open cockpit, wait for it to load the stream
3. Check webrtc-internals. 1 stream is running.
4. go to BlueOS, restart MCM
5. go back to cockpit, wait for the stream to connect
6. check webrtc-internals again, 2 new streams are running.

rafaellehmkuhl, 2024-03-07:
Interesting!

When you say the old one is finished, what do you mean exactly? It will keep consuming client resources, but not network bandwidth? Does it reflect somehow on the usage of the stream? Maybe connected to #775?
And on the MCM side, what's the outcome of this?

joaoantoniocardoso, 2024-11-12:
This is _very_ important for those running multiple cameras, as they'd immediately hit the connection bandwidth limit with 2 USB cameras. I just recorded it using the _video stats for nerds_ and BlueOS network widget, it's very easy to see the problem:

https://github.com/user-attachments/assets/c503889d-4322-452e-b823-d9587e4fe62e