bluerobotics/cockpit #815: [Video Recording] Events are not starting on the first video frame

joaoantoniocardoso, 2024-03-08:
QGC starts its events on the first video frame (time zero), I think we should also do that.

It might be a delay in the processing chain, or _when_ we are generating the timestamp -- the delay is not always the same.

Below is an image comparing two ass files, from Cockpit on the left, and QGC on the right:

[image]

rafaellehmkuhl, 2024-03-08:
Yes. 

What is probably happening is that, when we slice, we do not include sensor logging from a timestamp before the video start timestamp. What we can do is to always use the last sensor logging, that was received just before the video started, as the logging from 00:00:00 till the first register (in this case, 00:01:35). 

What do you think?

rafaellehmkuhl, 2024-03-08:
> Well, to be scientifically accurate, the timestamp should represent either the _start_ (the usual) or the _end_ (rare) of the sample. To insert data before 1:35 would be to extrapolate the measurements. We can either collect the data before the video starts, or we can choose not to follow QGC at all.

We are already capturing data before, we are just not adding it to the logging slice that goes to the subtitle.

joaoantoniocardoso, 2024-03-08:
Great, so we can do that :)

as in.. to insert the data sampled before the instant of the first frame and mark it as 0:00 until the next data sample arrives.

rafaellehmkuhl, 2024-03-08:
Yup. Marked as good-first-issue to open space for those interested in contributing.

closed 2025-02-06 by rafaellehmkuhl