bluerobotics/cockpit #832: Arming vehicle using action via joystick controller is not requiring slide to confirm

JoaoMario109, 2024-03-19:
When using the controller to perform actions that require confirmation as arm, disarm, etc... The confirmation pop-up isn't appearing. Now that we've implemented global enable/disable for slide to confirm lib in #761 and added the ability to confirm via the controller in  #822 we need to ensure that all actions requiring confirmation, such as arming and disarming, work with the controller, not just through the UI.

ES-Alexander, 2024-03-25:
Hmm, I don't think I agree with this.

Joystick buttons can be made intentionally hard to press accidentally (by physical design), and can also be composite actions already (by software configuration), in which case they shouldn't always need an additional confirmation step that requires removing hands from the joystick (e.g. like the "emergency stop" button on machinery, or using a [modifier key](https://github.com/bluerobotics/cockpit/issues/617) _as_ the confirmation of intent).

We should have a discussion on the desired behaviour here, because "take your hands off the joystick" doesn't seem like a practical default to me. If implementing nuanced behaviour is also impractical then my personal preference would be to encourage good practices when configuring the joystick profiles, rather than requiring confirmations off the joystick.

rafaellehmkuhl, 2024-04-01:
Just to confirm, the agreed solution to this is to allow the user to add the "hold to confirm" functionality to some actions, right?