bluerobotics/cockpit #863: When user changes to a profile with a VeryGenericIndicator, the config window opens and don't close

rafaellehmkuhl, 2024-04-05:
[image]

ArturoManzoli, 2024-04-05:
The issue is caused by the "MAV DEFAULT" profile having an unconfigured veryGenericIndicator on the bottom bar on its standard layout.

To replicate the issue:
* Clear the browser cache;
* Press the "Reset profiles to default configuration" button;
* Select the standard profile for MAV;
* The veryGenericIndicator config pops out. 

To overcome the problem:
* Just select a preset and then close the config popover.

Issue solution:
* The unconfigured veryGenericIndicator will be removed from the default layout for the "MAV STANDARD" profile;

[image]
[image]

rafaellehmkuhl, 2024-04-06:
Good catch!

Suggestion: to add a boot routine on the widget store that removes unconfigured VeryGenericIndicators. This way people which already have them (all that already have the MAV profile) won't face this bug.

PR #865 (2024-04-08): On profile change, unconfigured veryGenericIndicator pops and wont close
Changes prevent unconfigured veryGenericIndicators from loading in to the views on cockpit load or in profile changes

Fix #863
closed 2024-04-08 by rafaellehmkuhl