bluerobotics/cockpit #882: Controller SVG not appearing for unknown device in electron app on Windows

JoaoMario109, 2024-04-11:
When you clone and manually install the cockpit using bun on Windows, and run it in a browser, the controller is recognized with a default SVG, even if it's an unknown device. However, using the Windows electron app, the controller's SVG doesn't appear. 

Running on chorme (Windows):

[image]

Running on electron app (Windows)

[image]

Details from controller used in the test extract from gamepad tester.

[image]

rafaellehmkuhl, 2024-04-11:
It's not working also on Mac, and also for known controllers:

[image]

rafaellehmkuhl, 2024-07-03:
Update: it seems like the SVG cannot be fetched.

[image]

PR #1064 (2024-07-03): joystick-config: Use relative path for joystick SVG file
This way the standalone Electron app can also load it.

[image]

Fix #882
closed 2024-07-03 by patrickelectric