bluerobotics/cockpit #916: [WebRTC] There should be no stun/turn servers by default

joaoantoniocardoso, 2024-04-26:
### Proposed change

1. There should be no default stun/turn servers
2. The stun and turn servers should be configurable

### Why?

Although stun/turn servers are usually good because they provide additional ICE candidates, they are potentially routed through the internet (especially if using public stun servers like the ones from Google -- our current behavior), which can degrade significantly the perceived performance, adding latency and/or stuttering, depending on the quality of both connections (vehicle to internet, internet to Cockpit).

Relevant section of the code:
[image]

rafaellehmkuhl, 2024-04-26:
Agree.

@Williangalvani you were the one using those, right? How could we do this configuration in a way that fits your needs?

joaoantoniocardoso, 2024-04-26:
> Agree.
> 
> @Williangalvani you were the one using those, right? How could we do this configuration in a way that fits your needs?

...perhaps a BlueSim configuration?

rafaellehmkuhl, 2024-04-26:
> > Agree.
> > @Williangalvani you were the one using those, right? How could we do this configuration in a way that fits your needs?
> 
> ...perhaps a BlueSim configuration?

Yup. My point is how would Cockpit receive this. One way would be through query parameters, as will has implemented on PR #669.

closed 2024-04-30 by patrickelectric