bluerobotics/cockpit #917: [master] [WebRTC] 3 video players are being created for every stream

joaoantoniocardoso, 2024-04-26:
3 video players are being created for every stream

[image]

note: this problem doesn't exist on `v1.0.0-beta.14`

rafaellehmkuhl, 2024-04-26:
Didn't get it. It is on master but not on Beta14? The only PR we merged since was #910.

JoaoMario109, 2024-04-26:
The problem wasn't happening on the beta 12, as @joaoantoniocardoso was running in BlueOS and he had 2 versions running but by the UI that was running it was previously to the Beta 14

joaoantoniocardoso, 2024-04-26:
Ops, updated. 12 indeed

joaoantoniocardoso, 2024-04-27:
Useful for debug: `chrome://media-internals/`

joaoantoniocardoso, 2025-10-30:
This does happen: each player is from one of the views. 

This is probably not a problem per se, it only means that we might be player 3 times the same video on the browser. I don't know whether this uses more resources or not; it might be worth comparing the resource use.

rafaellehmkuhl, 2025-10-30:
Oh ok, this is different from what I thought was the problem initially.

I thought it was creating extra video players even with a single view and a single video widget.

In that case I will remove the Bug mark and rename the issue, as this is probably an optimization point instead.