bluerobotics/cockpit #933: Fix viewsToShow object contents reordering on view change

ArturoManzoli, 2024-05-06:
On view changes, the object returned by viewsToShow on the widgetManager store modifies the order of its contained views.
This is causing problems on the rendering of the selected view.

[image]

https://github.com/bluerobotics/cockpit/assets/14910201/f7ac9b63-866c-4e5a-80ad-24f3f937f9d3

rafaellehmkuhl, 2024-05-17:
I actually remembered why this is like this: we are using the order of this array to dictate which view should be on the top (hence be shown). Will try to figure out a way to change it.

rafaellehmkuhl, 2024-06-19:
@ArturoManzoli I'm thinking if this is really a problem.

The issue of being able to click the Map buttons when on another view, I couldn't reproduce, and the iFrame problem we got to know it's not related to that.

What do you think?

rafaellehmkuhl, 2024-06-27:
For the moment I'm closing this issue as stale as it is apparently not affecting Cockpit in any bad way.