bluerobotics/cockpit #948: Clicking to download video caused Cockpit to offer a zip with the chunks

rafaellehmkuhl, 2024-05-10:
That was a problem faced by Brian Austin (BR):

> When I clicked the large download icon in the bottom right, it downloaded a large zip file of 326 files in it and none were the actual video file.  Then I was able to click on the menu on the actual video and download from there and that one worked and gave me the video file

ArturoManzoli, 2024-05-14:
I haven't experienced that as an issue or bug yet, but I do recognize this behavior as intentional.
What might have happened to Brian is one of the following situations:

1- If you select an unprocessed video, the download will be a ZIP containing the video's chunks. Downloading the chunks could be useful if the cockpit cannot process the video.
2- If you select multiple unprocessed videos, the download will be a ZIP containing all selected video chunks in the root directory, with files named by hash.
3- If you select multiple videos and among them there are both processed and unprocessed videos, the download will consist of two ZIP files: one for the processed .webm and .ass files, and another for the unprocessed ones.

rafaellehmkuhl, 2024-05-14:
Got it.

So it's an UX problem, not a bug in the software.

Maybe a solution is to add a confirmation dialog saying that the user is downloading chunks instead of the video, on those situations?

rafaellehmkuhl, 2024-05-14:
> I haven't experienced that as an issue or bug yet, but I do recognize this behavior as intentional. What might have happened to Brian is one of the following situations:
>
> 1- If you select an [...]

Got it.

So it's an UX problem, not a bug in the software.

Maybe a solution is to add a confirmation dialog saying that the user is downloading chunks instead of the video, on those situations?

ArturoManzoli, 2024-05-14:
Maybe we should also differentiate more clearly the processed and unprocessed videos on the Video Library.

On top of that, we can change the ZIP filenames for something more descriptive. 
Now processed videos are named `Cockpit-Video-Recovery - 2024-05-14T094324.981.zip` and unprocessed are named  `Cockpit-Temp-Video-Chunks-88f8f692.zip`

rafaellehmkuhl, 2024-05-14:
More info on that: Brian said he had clicked the process button already. Maybe the download button was available while the video was being processed?

ArturoManzoli, 2024-05-14:
Yes, this might have happened. The button should disabled while the videos is processing.

closed 2024-05-23 by rafaellehmkuhl