bluerobotics/cockpit #951: bug: deleting videos isn't working

ES-Alexander, 2024-05-14:
https://github.com/bluerobotics/cockpit/assets/25898329/34e2e743-9d3d-471b-a340-76ee714ee5f3

[image]

ES-Alexander, 2024-05-15:
I would likely never guess that that’s supposed to be a button - it looks like a status indicator (like “sure - I have deleted those things for you”), especially since it’s freestanding and located partway on top of the video preview, away from the other buttons.

> Do you have any suggestions to make this process more intuitive, but keeping the UI minimalist (no confirmation modals)?

The simplest approach would be to have no confirmation (one click delete), but if you want confirmation without a modal then I’d recommend the first click changing the delete button into either the green tick button or (more clearly) a “confirm delete?” button (possibly with an added “X” or “cancel deletion” button to the left), so it’s obvious you’re in a second stage of the “attempting to delete” process :-)

rafaellehmkuhl, 2024-05-15:
> The simplest approach would be to have no confirmation (one click delete), but if you want confirmation without a modal then I’d recommend the first click changing the delete button into either the [...]

Transforming the delete button on a confirmation button seems very cool indeed!

PR #955 (2024-05-17): Improve video deletion UI
* Added extra conformation button on Video Delete;
* Modified confirmation buttons positions and animation;

Fix #951
closed 2024-05-17 by rafaellehmkuhl