bluerobotics/cockpit #992: VeryGenericIndicator Mini Widget - Would be more generic if custom mode just allowed you to format using a template

pkmiles, 2024-06-09:
In attempting to add Lat, Lng onto the footer bar I currently have to do as two seperate variables. Also, having done this I can't easily get the formatting at the right display precision.  

Given Custom mode is a power feature what if you went away from the set variable picker to a textarea field where you could enter your format using a moustache/handlebars template for variables and formatting?

That would allow something like the attached example. 
 

[image]

rafaellehmkuhl, 2024-06-09:
That's an interesting idea actually...

In this case we should allow the VGI to auto-scale in width also.

Let me do some tests.

ES-Alexander, 2024-07-01:
Relevant to #613