bluerobotics/cockpit #1007: speechSynthesis.speak() without user activation is deprecated and will be removed.

joaoantoniocardoso, 2024-06-15:
[image]

rafaellehmkuhl, 2024-06-16:
Humnnn

We probably want to watch for vehicle arming and offer the user a button to activate the speech during the mission. It would actually be better than what we have today, where the user opens Cockpit and all of a sudden the speech person starts to shout.

@ES-Alexander @ArturoManzoli what do you guys think?

ES-Alexander, 2024-06-16:
> We probably want to watch for vehicle arming and offer the user a button to activate the speech during the mission.

This could potentially work, if we had a setting for it with "turn on" / "ask me later" / "don't ask me again" options.

That said, Cockpit should likely have a wizard or similar on first connection, and setting that up as part of that process seems quite logical as well. Ideally the initial configuration could be done by opening and highlighting components of the standard interface, so by doing the setup the user also learns about the different sections and major features of the application.[^1]

Something like:
1. Connect to a vehicle? 
1. Upload a custom profile?
2. Select current view
3. Select camera stream?[^2]
4. Choose generic indicators?
5. Configure notifications (importance level threshold + speech settings)
6. Configure logging
7. Calibrate / configure joystick[^2]
8. Import custom MAVLink/JS functions[^2][^3]
9. Record and download video[^2]
10. Set level horizon[^3] / point to BlueOS sensor calibration
11. Arm vehicle
12. Disarm vehicle

[^1]: Speaking of, that kind of approach could be worth considering for BlueOS as well...

[^2]: Optional, depending on what's connected

[^3]: Feature not currently available

rafaellehmkuhl, 2024-06-17:
> > We probably want to watch for vehicle arming and offer the user a button to activate the speech during the mission.
> 
> This could potentially work, if we had a setting for it with "turn on" / "ask me later" / "don't ask me again" options.
> 
> That said, Cockpit should likely have a wizard or similar on first connection, and setting that up as part of that process seems quite logical as well. Ideally the initial configuration could be done by opening and highlighting components of the standard interface, so by doing the setup the user also learns about the different sections and major features of the application.[1](#user-content-fn-1-c0c8bd389a787cbe92aeefe51b322c57)

I agree that Cockpit should have a wizard at some point. Would you mind opening another issue with what you suggested here?

Regarding the speech problem, it actually demands that the user activates it on every Cockpit usage. That's why I was thinking in something linked to the vehicle arming.

ES-Alexander, 2024-06-17:
> Would you mind opening another issue with what you suggested here?

Done.

> Regarding the speech problem, it actually demands that the user activates it on every Cockpit usage. That's why I was thinking in something linked to the vehicle arming.

Oh, that's kinda gross, but ok, if there's no way to make it a persistent setting then I agree that a pop-up when they first try to arm in a session is probably a reasonable compromise