bluerobotics/cockpit #1008: Possibly wrong warning appearing when recording video

joaoantoniocardoso, 2024-06-15:
My system seems perfectly fine (~10% CPU use, ~6% Ram), but Cockpit is warning whenever I record a video (v1.0.0-beta.20)

https://github.com/bluerobotics/cockpit/assets/5920286/8eb1ca26-4ca7-49d1-8390-024d99947298

rafaellehmkuhl, 2024-06-16:
Will investigate.

rafaellehmkuhl, 2024-06-16:
@joaoantoniocardoso I can't reproduce the problem here.
Can you get us a Cockpit log from that session?

rafaellehmkuhl, 2024-07-09:
@joaoantoniocardoso I could not reproduce this problem and talking to Tony, Eldin and Will, none of them have experienced this problem, so I assume there was indeed a problem when you received this warning.

Will close for now. If you happen to have the problem again, feel free to reopen the issue.

closed 2024-07-09 by rafaellehmkuhl