bluerobotics/cockpit #1047: Memory leak in cockpit

patrickelectric, 2024-06-28:
[image]

patrickelectric, 2024-06-28:
This does not happen when the vehicle is not connected

rafaellehmkuhl, 2024-06-28:
In the standalone version I could also reproduce a leak, but not in the same magnitude. I got the following values:
- 0min: 190mb
- 10min: 280mb
- 20min: 390mb
- 30min: 580mb

joaoantoniocardoso, 2024-06-28:
@rafaellehmkuhl seems exponential growth?
[image]

rafaellehmkuhl, 2024-06-28:
> @rafaellehmkuhl seems exponential growth?

In my tests not, but in Patrick's yes.

I'm going to investigate that better.

PR #1066 (2024-07-04): ardupilot: Do not duplicate path of available generic variables
Left Cockpit running for an hour an it didn't reach 200MB (without a camera connected).

Before:

https://github.com/bluerobotics/cockpit/assets/6551040/b8348dc5-8eb4-4128-b687-112480a17b4b

After:

https://github.com/bluerobotics/cockpit/assets/6551040/e5204aa3-9c5d-4d71-9190-f57f5c80c01e

Fix #1027 
Fix #1047
closed 2024-07-04 by rafaellehmkuhl